cornucopia-machines/ugly-duckling-firmware #26: Bad water-measurement

doramelindatunde, 2024-01-02:
Mains should only be used to measure drip-losses in Grafana, otherwise we get duplicated data.
Thus, the measurements of all boxes after mains should be subtracted. We should keep in mind that linearly timed watering will allow much fewer measurement points (mains, drains, animals) while still giving good water/bed data.

lptr, 2024-02-05:
Do we want to do something here still?

closed 2024-07-17 by lptr

doramelindatunde, 2024-08-09:
ok

lptr, 2024-08-16:
I've started measuring the different kinds of water flow meters. They have quite different characteristics based on the amount of flow.

Measurements:

https://docs.google.com/spreadsheets/d/1AdiQDayEqgdzuvxyAnY5iqzs-Qx-oSkFT-AGj4IGXWo/edit